goodrobots/vision_landing #13: Add ROS adapter instead of vision_landing script

fnoop, 2017-01-24:
vision_landing python script is essentially a wrapper to track_targets that connects to the flight ontroller using dronekit.  Another option is to connect track_targets with ROS and then consume the vectors through mavros.

TSC21, 2017-11-08:
This already exists, though WIP on my side - https://github.com/mavlink/mavros/pull/311

fnoop, 2017-11-08:
px4 is struggling to get it implemented in firmware..  Will your mavros solution work for arducopter?

TSC21, 2017-11-08:
> px4 is struggling to get it implemented in firmware

It's a matter of time. Probably it will be me that is going to generalize it to `LANDING_TARGET`.

> Will your mavros solution work for arducopter?

It should.

kripper, 2023-03-16:
> This already exists, though WIP on my side - [mavlink/mavros#311](https://github.com/mavlink/mavros/pull/311)

@TSC21 what are you using for the computer vision part (tracking markers)?