goodrobots/vision_landing #34: Set limits and constraints on tracking

fnoop, 2017-02-20:
patrickpoirier51 @patrickpoirier51 12:18
@fnoop Concerning the runaway behavior, may I suggest that for the moment, we set a limit on the tracking range ( -0,2 to 0,2) and return to zero if no reading more than N millisecond (300-500 max)? Here is an an example if the system loosing tracking for a few seconds = on a live system that is equivalent of pushing sticks to extreme position !!

patrickpoirier51, 2017-02-20:
Concerning this issue, AC_PrecLand sets the Target Acquired here : https://github.com/ArduPilot/ardupilot/blob/master/libraries/AC_PrecLand/AC_PrecLand.cpp#L205

Look at the number, its  up to 2000 msecs !! 
I will use this parameter with a lower value (like 300-500 msec)  instead of adding an additionnal timer within vision_landig code.  But still need to implement a simple limiter until we get the EKF timing issue resolved.

fnoop, 2017-02-20:
@patrickpoirier51 Actually I thought your idea of returning 0,0 if outside of time threshold was really great!  So within the python code if we don't get a target fix from the vision within the last 250ms or so, then just return 0,0 (with the last Z value I guess).  This is a quick fix as well for AC_Precland not handling losing target well, as it would think it still has the target.

patrickpoirier51, 2017-02-20:
@fnoop  Yeah lets stick with it, ithe Target Acquired is just looking for the presence of a message but  not the quality of the message.

fnoop, 2017-02-20:
Hmm actually thinking about it that's not going to work.  Returning 0,0 from the landing_target means different things depending on the attitude of the craft, we're effectively telling it the landing target is always directly underneath below the craft even if it's pointing sideways or upside down.  It would be nice if we could send NED coordinates from the pose estimation but that would bring it's own problems and complications.
This should be fixed from within AC_PrecLand.

closed 2017-02-20 by fnoop