goodrobots/vision_landing #76: PX4 Compatibility

petergerten, 2017-07-03:
As the system uses MAVLink for communication, I assume it might also work with the PX4 flight stack?
Has anyone tested this?

fnoop, 2017-07-03:
Good question :)  No idea, I've never used px4, but it's on my todo list to try.  Will keep this open as a reminder.

TSC21, 2017-07-03:
PX4 still doesn't support `LANDING_TARGET` msg decoding and usage, so that's a no. I'm working on that at the moment.

fnoop, 2017-07-03:
@TSC21 Are you working on coding support for LANDING_TARGET in px4 firmware?  Are you planning on doing something more complex than sending camera frame angles (https://github.com/mavlink/mavlink/pull/744)?

TSC21, 2017-07-03:
Yes I am. Coding it right now according to https://github.com/mavlink/mavlink/pull/744. Till it is merged and applied on `mavlink_gdb_release`, I'm working locally to make it work.

fnoop, 2017-07-03:
Why do you need to extend landing_target?  Can you not code px4 to use camera frame radians?

TSC21, 2017-07-03:
> Why do you need to extend landing_target? Can you not code px4 to use camera frame radians?

I could but it much more intuitive to use full pose of the target instead (it may happen that the landing target isn't even in a flat surface, for example).

fnoop, 2017-07-03:
Sounds interesting, I agree it's much nicer to use the pose if it's available.  Bonus points if you can support a gimbal :)

TSC21, 2017-07-03:
That can also be considered. In ROS you can listen to transforms, and knowing the current camera pose, you can compute the transform from the target (which will be in a certain frame - better to consider NED frame as we already do the transforms from ENU to NED), to the camera center. That way you don't really need to have the camera always pointing down, and you can basically have it tracking the the landing target. But you will need a mechanism that gives you feedback of the current gimbal position (I suppose that if you have a SToRM32, you can have a way of getting the gimbal orientation and fed it to the position controller, or even to ROS, if you had some kind of msg that could send that info).

fnoop, 2017-07-03:
Even if the flight controller only controls the gimbal through PWM, it should be able to use the intended position instead of actual position through feedback from the gimbal/imu?

TSC21, 2017-07-03:
Well that's how controllers and actuators work I suppose: you know the orientation you want to give to the camera (for example, you wanna fix the position of the target so you can fix the gimbal on it), so you use the controller to compute the desired orientation to it. http://ardupilot.org/copter/docs/common-storm32-gimbal.html here you have a way of knowing what values to send based on the max angles and rates.

TSC21, 2017-07-03:
I'm picking the SToRM32 cause it is Mavlink-based and with neat features.

petergerten, 2017-07-05:
@TSC21 wow, great that you are working on it. I am happy to do testing :)

petergerten, 2017-07-22:
@TSC21 did you get further with this? I could not find exactly figure out what the state of it is in the px4 source.

TSC21, 2017-07-22:
Still working on it

petergerten, 2017-07-22:
I am a bit confused on where this would actually be implemented. Will this be part of the core px4 firmware? I saw MAVROS sugested here [http://discuss.px4.io/t/visual-precision-landing/2418/9](http://discuss.px4.io/t/visual-precision-landing/2418/9)

TSC21, 2017-07-22:
That can be an approach also. But to be available, it will be on the firmware side. MAVROS serves as a bridge, not as an app, so you have to write your own node to do that. Still ROS would be useful to retrieve and process marker tracking, while the landing process can be implemented om both sides.

petergerten, 2017-07-22:
@TSC21 thanks for the explanation. Can you give me a hint on where you implement the feature in the px4 source? Is that a public repository or will you only publish it when it's "finished"?

TSC21, 2017-07-22:
I'm going for the ros implementation for now as it is quicker. For the px4 implementation could take some more time as the `navigator` and `commander` modules are in a reformulation process from the core.

petergerten, 2017-08-06:
@TSC21 how is the ROS implementation coming along? :)

petergerten, 2017-10-02:
@TSC21 Can you give me a hint on where to look/start? Will this be in the mavros repo? https://github.com/mavlink/mavros/commits?author=TSC21

petergerten, 2017-11-02:
@TSC21 Any pointer on where to look would be great. I can't find the related code anywhere. Or are you currently doing the development in private?

TSC21, 2017-11-02:
https://github.com/PX4/Firmware/pull/8160

fnoop, 2017-11-02:
@TSC21 Good to see some progress.  It's not entirely clear from that PR conversation if they have gone the generic route, or if it's irlock only as per the original PR?

TSC21, 2017-11-02:
For now is just the irlock. If Nicolas does not just generalize it, I will do it myself.

petergerten, 2017-11-03:
thanks. Indeed very good to see some progress. Hoping for a generic implementation and not the irlock specific.

fnoop, 2018-01-18:
@TSC21 Now irlock precland support is in px4, how do we get mavlink (or more likely rtsp) sensor support in?

TSC21, 2018-01-18:
I am working on that but I am currently swamped with other matters.

petergerten, 2018-07-13:
I wonder if the new VIO stack for PX4 resulted in any improvements that allows more generic precision landing implementations with PX4 now (rather than only irlock).

fnoop, 2018-07-13:
@petergerten That would be great, I'm resurrecting this project hopefully with proper support in ardupilot, would be great to support px4 as well.
@TSC21 Any idea if we can do this now in px4?

TSC21, 2018-07-13:
It already supports external vision landing yes. But it expects the full pose and not angles.

fnoop, 2018-07-13:
Are there any docs or sample code?  I couldn't see any at a quick search.

TSC21, 2018-07-13:
You did commented on it as well: https://github.com/PX4/Firmware/pull/9050

petergerten, 2018-10-25:
Hi @fnoop @TSC21
Can I use this with PX4 vision landing now or what is missing at the moment?
Above you said something regarding full pose requirement for LANDING_TARGET.

petergerten, 2019-05-25:
I am still very interested in getting it running with PX4 or PX4+ROS - did anyone get closer on this yet?

TSC21, 2019-05-25:
@petergerten the implementation is done as you can see above. The only thing is that the landing_target plugin was not merged yet, though it is usable (I am exploring it in 2 different projects currently). The only thibg is that the data that you send on thr message needs to be the absolute positiom of the target on the local frame, since the current implementation in PX4 expects that (which is not completely right). Though I do consider that this issue can be closed, since the solution is already present.

alchemyconsulting, 2019-05-25:
@TSC21 Not even remotely 'closed' issue - px4 support is unuseable to anyone that isn't 'in the know', at least last time I looked.  If you can point to documentation or guide how to use it, then maybe the rest of the world can use it.
I think developers often get too involved in what they're doing, and forget that technology needs to be useable by end users, in order to really exist at all.   If it's not available to others, it might as well not exist.  We're years down the line and still this isn't a widely used technology which is crazy, as it was so close in both ardupilot and px4 a long time ago.

petergerten, 2019-05-25:
@TSC21 You mentioned "The only thibg is that the data that you send on thr message needs to be the absolute positiom of the target on the local frame" -> how would I do that?

@alchemyconsulting I fully agree with your sentiment...

TSC21, 2019-05-25:
@alchemyconsulting if you feel is not well documented or not well supported, then this is something to be discussed and adressed in the appropriate PX4 or issue topic, not here. Also we do have documentation for it: https://docs.px4.io/en/advanced_features/precland.html. though it specifies the usage of the IR Lock only, a close look up on the link I sent in some comment above will lead you to understand that also external vision landing target estimatiom feeding is also possible. This is though not  matter that is on the top of the priorities at the moment for the dev team in PX4 and not something I do have time to explore the development.

@petergerten again, if you want this to be something to be explored and answered, this is not the best place to do it. Just bring this to the PX4 forums ;)

petergerten, 2019-05-25:
@TSC21 thanks. I will bring in up in the px4 forums. Can you clarify which branch / pull request you are talking about that you are already using but is not merged yet?

TSC21, 2019-05-25:
If you read closely what I wrote above, you will see that the PR to bring support for LANDING_TARGET msg is already merged - PX4/Firmware#9050. What's not merged is the PR to MAVROS to add the landing_target plugin - https://github.com/mavlink/mavros/pull/311 - , but that's PX4-independent.

petergerten, 2020-02-21:
@TSC21 Would it also be possible to use the path planning interface to facilitate a landing in mission mode and just send TRAJECTORY_REPRESENTATION_WAYPOINTS messages similar to the safe landing planner?