goodrobots/vision_landing #77: Indoor Loiter

anbello, 2017-07-19:
Sorry if this is not the right place to make this question,
I would like to know if it is possible to use aruco marker pose estimation as a form of visual odometry for indoor navigation. Or at least to keep a drone hovering in a fixed position without GPS.

fnoop, 2017-07-20:
@anbello Hi :) . You can absolutely use this (or Ardupilot precland in general) to keep a drone hovering in fixed position without gps.  You need to use 'precision loiter' for this, if you look at the wiki page:
http://ardupilot.org/copter/docs/precision-landing-with-irlock.html
It says:
```
To enable Precision Loiter, an Auxiliary Function Switch must be set to “Precision Loiter”:
CH7_OPT = 39 (or alternatively set CH8_OPT to CH12_OPT)
```
@patrickpoirier51 has successfully done this in his 'Thunderdrone' garage.

Visual Odometry is a completely different kettle of fish, and is a work in progress in Ardupilot.  For VO/localisation using aruco targets, you might be interested in this:
http://www.uco.es/investiga/grupos/ava/node/57

TSC21, 2017-07-20:
@anbello if you are using ROS, maybe this can interest you - http://wiki.ros.org/ar_sys. You can basically then send the position estimate to Ardupilot to be fused on EKF3.

closed 2017-09-11 by fnoop