goodrobots/vision_landing #81: vision_landing kills track_targets on Raspberry

fnoop, 2017-08-20:
```
2017-08-20 14:37:34,980 INFO Launching track_targets with arguments:/srv/maverick/software/vision_landing/track_targets -d TAG16h5 -w 640 -g 480 -f 30 -z 11:0.446,16:0.204,19:0.104,23:0.047 --markerhistory 15 --markerthreshold 90 /dev/video0 /srv/maverick/software/vision_landing/calibration/raspicamv1-calibration-640x480.yml 0.5
2017-08-20 14:37:34,997 INFO Connecting to drone on: tcp:127.0.0.1:5780
2017-08-20 14:37:36,919 INFO Connected to drone
2017-08-20 14:37:36,920 INFO Setting flight controller parameters
2017-08-20 14:37:36,935 INFO Entering main tracking loop
2017-08-20 14:37:36,936 INFO Requesting track_targets to start tracking:VehicleMode:RTL
2017-08-20 14:37:37,995 ERROR Error: Tracking process shut down unexpectedly (-10), restarting
2017-08-20 14:37:37,996 INFO Launching track_targets with arguments:/srv/maverick/software/vision_landing/track_targets -d TAG16h5 -w 640 -g 480 -f 30 -z 11:0.446,16:0.204,19:0.104,23:0.047 --markerhistory 15 --markerthreshold 90 /dev/video0 /srv/maverick/software/vision_landing/calibration/raspicamv1-calibration-640x480.yml 0.5
2017-08-20 14:37:39,041 INFO Requesting track_targets to start tracking:VehicleMode:RTL
2017-08-20 14:37:39,546 ERROR Error: Tracking process shut down unexpectedly (-10), restarting
2017-08-20 14:37:39,551 INFO Launching track_targets with arguments:/srv/maverick/software/vision_landing/track_targets -d TAG16h5 -w 640 -g 480 -f 30 -z 11:0.446,16:0.204,19:0.104,23:0.047 --markerhistory 15 --markerthreshold 90 /dev/video0 /srv/maverick/software/vision_landing/calibration/raspicamv1-calibration-640x480.yml 0.5
2017-08-20 14:37:40,625 INFO Requesting track_targets to start tracking:VehicleMode:RTL
2017-08-20 14:37:41,178 ERROR Error: Tracking process shut down unexpectedly (-10), restarting
2017-08-20 14:37:41,179 INFO Launching track_targets with arguments:/srv/maverick/software/vision_landing/track_targets -d TAG16h5 -w 640 -g 480 -f 30 -z 11:0.446,16:0.204,19:0.104,23:0.047 --markerhistory 15 --markerthreshold 90 /dev/video0 /srv/maverick/software/vision_landing/calibration/raspicamv1-calibration-640x480.yml 0.5
2017-08-20 14:37:42,215 INFO Requesting track_targets to start tracking:VehicleMode:RTL
2017-08-20 14:37:42,343 ERROR Error: Tracking process shut down unexpectedly (-10), restarting
```

It looks like track_targets is shut down and returns -10 return code, which corresponds to being killed by sig 10 (SIGUSR1).  

vision_landing uses SIGUSR1 to command track_targets to wake up and start processing:
```
    def start(self):
        log.info("Requesting track_targets to start tracking:"+str(craft.vehicle.mode))
        self.state = "started"
        self.process.send_signal(signal.SIGUSR1)
```
When run by hand outside of vision_landing master process, it seems to work fine:
```
[dev] [mav@maverick-raspberry ~/var/log/vision/vision_landing]$ /srv/maverick/software/vision_landing/track_targets -d TAG16h5 -w 640 -g 480 -f 30 -z 11:0.446,16:0.204,19:0.104,23:0047 --markerhistory 15 --markerthreshold 90 /dev/video0 /srv/maverick/software/vision_landing/calibration/raspicamv1-calibration-640x480.yml 0.5
info:FoVx~51.8032:FoVy~40.1442:vWidth~640:vHeight~480
info:Size Mappings:11=0.446, 16=0.204, 19=0.104, 23=0.047,
debug:Marker History:15
debug:Marker Threshold:90
SIGNAL:SIGUSR1:Received:10:Switching on Vision Processing
```

vgroove, 2017-09-05:
Got it working on the pi by adding two things.  The first by adding a boolean to make sure start() and stop() only get called once, since when controlprocessing=False it flooded the subprocess with SIGUSR1.

In main loop in visionlanding
```python
tracking = False
while True:
    # If craft mode has changed, take action
    if (args.controlprocessing and (craft.vehicle.armed and (craft.vehicle.mode == "LAND" or (craft.vehicle.mode == "LOITER" and (craft.precloiter_opt and craft.vehicle.channels[str(craft.precloiter_opt)] > 1800))))) and track_targets.state != "started" and not tracking:
        track_targets.start()
        tracking = True
    elif (args.controlprocessing and (not craft.vehicle.armed or (craft.vehicle.mode != "LAND" and (craft.vehicle.mode != "LOITER" and (craft.precloiter_opt and craft.vehicle.channels[str(craft.precloiter_opt)] <= 1800))))) and track_targets.state == "started" and tracking:
        track_targets.stop()
        tracking = False
    elif not args.controlprocessing and not track_targets.state and not tracking:
        track_targets.start()
        tracking = True
```

and the second was by adding a sleep right after track_targets is launched to make sure that initial start signal gets to it.  

In TrackTargets.__init__
```python
# Launch tracking process and process output
log.info("Launching track_targets with arguments:" + " ".join(self.track_arguments))
self.launch()
log.info("Sleeping for 10 seconds...")
sleep(10)
```

Didn't make a PR because there is almost definitely a more elegant way of doing this, but maybe this will help.

fnoop, 2017-09-11:
Revamped the tracking object and track_targets output to track the state better, including initialisation.  This issue was related to slower platforms taking a while to start up the track_targets process and vision_landing sending the SIGUSR1 before the signal handling in track_targets was setup.  Default handling of SIGUSR1 is to terminate
Instead, track_targets now sends a specific message once initialised which vision_landing waits for, before trying to send the signal to start tracking.

closed 2017-09-11 by fnoop