goodrobots/vision_landing #94: Move to either c++ or python, away from dual/hybrid model

fnoop, 2018-07-11:
vision_landing is currently a python wrapper/supervisor that talks to dronecode, and calls c++ code for the visual processing.
This works, but turns out to be overly complex to coordinate the two halves, and dronecode is very slow/inefficient.
As we want to get vision_landing running on much slower platforms -ie. raspberry - migrate to:
 - single code, either python or (probably) c++
 - away from dronecode, to either direct mavlink or mavros connector

fnoop, 2023-03-16:
> C++ is faster
If python is fast enough for the job, it's a much more friendly language to code a project like this in, for others who want to experiment/contribute.  My personal opinion is c++ should only be used where necessary for uses outside of the main flight controller code, because the number of people who can safely write not-to-buggy-code in c++ is much lower.

kripper, 2023-03-16:
Agree.