goodrobots/vision_landing #105: Could you give 1 more line of description in your warning?

bu-ra3y, 2018-08-07:
You write:

> ArduPilot and PX4 autopilots do not yet safely support vision based precision landing. 

Your project is utterly compelling. Why do you say the above? Are you saying that the shortcoming is in the autopilots? I would have guessed that it was your own project that needed more time and work, but you are pointing to the autopilot software. Is that intentional? If so, as a follower of your project i'd love to understand 1 more line or so of context about how the autopilot software limits your project. 

GOOD LUCK!

fnoop, 2018-08-09:
Hi @bu-ra3y!  Glad you like the project!  So the warning about the autopilot is down to limitations in the autopilot software, so far.  When I started this project, PX4 firmware did not support precision landing at all.  There is now some preliminary (but not complete) support for precision landing, which I will look into shortly.
Ardupilot is a little more complex - Precision Landing has been supported for a while, to varying degrees, but it is not very complete or safe to use at the current time.  It is tuned and designed around the hardware IRLock system, which uses InfraRed beacons and a very specifically tuned camera vision system to look for the beacons.  The more generic 'Companion' backend to Ardupilot PrecLand takes Mavlink messages instead which is what this vision_landing project uses to communicate with the autopilot.  However, the system does not cope well with longer and varying latencies of the messages that is typical with general computer-based vision systems, and this leads to unpredictable and sometimes quite unsafe behaviour (rapid movements into the nearest wall/garage/tree/person).

I am in the middle of updating Ardupilot to work better with more generic computer vision systems such as vision_landing, but it will take a little while to complete and get merged and then released:
https://discuss.ardupilot.org/t/ac-precland-improvements-and-updates/31614

I hope to start testing with Raspberry and Up Core boards in the next few days, which have much higher latency and jitter than previous tests.

bu-ra3y, 2018-08-10:
Love this.