goodrobots/vision_landing #117: global name 'timespec' is not defined

aydal, 2020-01-22:
Running vision_landing, I get this error. Earlier I got monotonic_time not defined, which I corrected by shifting the function to top. How to rectify this?

adrianpang, 2020-03-11:
ubuntu@M5510:~/Downloads/vision_landing$ ./vision_landing
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "./vision_landing", line 43, in insertQueue
    line = pipe.readline().rstrip() + ":{:.0f}".format(monotonic_time())
NameError: global name 'monotonic_time' is not defined

Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "./vision_landing", line 43, in insertQueue
    line = pipe.readline().rstrip() + ":{:.0f}".format(monotonic_time())
NameError: global name 'monotonic_time' is not defined

ubuntu@M5510:~/Downloads/vision_landing$ python --version
Python 2.7.17

Please see above for details on this issue.

fnoop, 2020-03-12:
OK a blank ubuntu 18.04 vm install, I can reproduce this:
```
~/vision_landing$ ./vision_landing
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "./vision_landing", line 43, in insertQueue
    line = pipe.readline().rstrip() + ":{:.0f}".format(monotonic_time())
NameError: global name 'monotonic_time' is not defined

Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "./vision_landing", line 43, in insertQueue
    line = pipe.readline().rstrip() + ":{:.0f}".format(monotonic_time())
NameError: global name 'monotonic_time' is not defined
```
Thanks for reporting, will fix!

fnoop, 2020-03-12:
Addressing outdated install instructions in #118